Jackysi/advancedtomato #300: Tenda N60 - Update to latest version of Advanced Tomato - The file is too bit to fit in MTD.

gxgl, 2017-02-26:
Hi,

I have tried to install the latest advancedtomato firmware (listed here: https://advancedtomato.com/downloads/router/n60 ) but I get the error from the title. As I read on the internet that file should fit in 8MB. As it shows on my computer is 7.85MB. Then I tried the latest version of tomatousb file which is 7.68MB as is shown on my pc and it worked. 

I have tried many times to get it installed but failed as described. I have erased nvram, I have tried also after uploading the init image file. 

What I think is, the file is a little bit over sized, but I wait for your answer. Also I not exclude the fact that maybe I do something wrong in the update/upgrade process.

Thank you,
George

gxgl, 2017-02-26:
Hi,

looks like the last image that worked is Version 3.2-137 which is 7.71MB. Those that are over this size didn't work giving the "The file is too big to fit in MTD" message and rebooting.

Thank you,
George

darkluk13, 2017-03-21:
Yea, i opened issue with this problem while ago, when 1st rom with this arive, but its still same after another update :(

Jackysi, 2017-03-21:
I can't reduce the firmware more because I don't have the knowledge to do so. I already removed what I knew wouldn't hurt users too much. Since I can't avoid this, I will remove support for N6. I will soon add message about deprecation and next release won't have the N6.

kaizerlord, 2017-07-27:
Hello, i try a upgrade today with the new version and in the progress appear that error, the file is to big, i will erase nvram and try to restore but dont upgrade firmware only reboot a router, any solution for this issue?

eaglgenes101, 2017-10-10:
Can also confirm that file is too big to blindly load onto the router.

closed 2025-06-02 by gxgl