Jackysi/advancedtomato #304: R7000 Device List / Static DHCP list

SanderSchutten, 2017-03-18:
Hi,

I love using AdvancedTomato v3.4-r138 on my R7000. I've disabled the WLAN radio's and use it as a router behind a set of Ubiquti Unifi access points. Thanks for developing the user interface.

I think there is an issue when there are many devices on the Device List or the Static DHCP list. When opening these pages the loader at the top of the screen goes from left to right (indicating that it's loading), but responds then with an error that the device is not accessible. I'll investigate the issue further when it arises again. 

Just wanted to report the issue already so others may recognize the issue. 

Keep up the good work. I check on a regular basis if a new version is available...

thanks,
Sander

Jackysi, 2017-03-21:
It could be IPv6 issue if you are using it?

SanderSchutten, 2017-03-27:
I have IPv6 disabled on the router. The issue hasn't arisen any more in the meantime.

closed 2017-03-27 by SanderSchutten