Jackysi/advancedtomato #306: The field "wan_modem_ipaddr" is invalid. Please report this problem.

mstarks01, 2017-03-29:
I tried to get OpenVPN working and ever since then, have had strange issues. For example, on the Network screen, when saving, I get "The field "wan_modem_ipaddr" is invalid. Please report this problem." Incidentally, there doesn't seem to be a problem with the WAN IP.

The router had reverted from Access Point + WDS back to only Access Point and, due to this error, I cannot switch it back.

mstarks01, 2017-03-30:
...and the router reset itself back to default settings when it was rebooted. Frustrating. I restored from backup and am now back to the original problem.

mstarks01, 2017-03-31:
I upgraded to the latest release and now it's working.

closed 2017-04-13 by Jackysi

mwaterbu, 2018-02-21:
I have run into this same issue (after setting up OpenVPN, seeing this and other similar errors). I believe this is due to NVRAM being full (specifically from a large amount of text for the "Certificate Authority" key). There may not be enough protection against filling up NVRAM, maybe causing settings to be overwritten and/or lost.
I am using an RT-N16 which only has 32kb of NVRAM. I have since freed up some space by removing QOS and port forwarding rules, but the error is not corrected.

I think I will need to reflash to correct this, as the extent of corrupted settings is unknown.