Jackysi/advancedtomato #307: QoS broken

simonz21, 2017-04-04:
I've been using the latest Shibby up till now and have only just switched over to AT. But there is a problem. On my N16 QoS no longer works. It looks like nearly all my traffic is being labelled as the default classification P2p/bulk and so I'm only getting a fraction of my bandwidth, instead of the 70% ceiling I have actually fixed

Jackysi, 2017-04-13:
Did you clean NVRAM?

simonz21, 2017-04-14:
Yup…sorry, went back to SHibby in the end…

Kazula, 2017-04-15:
I can remember that there was an entry in qos-classify.asp which mentioned UDP 1-65535 as Crawl P2P.
Lately, Google (also Youtube and other Google services) sends their data via QUIC (UDP) so that I think that the data gets classified as P2P/Bulk.

You could try to block the port 443 for UDP in #advanced-restrict.asp to disable QUIC.

closed 2017-04-25 by Jackysi