Jackysi/advancedtomato #311: PPTP Client Password not quoted

alanruttenberg, 2017-04-19:
The contents of the password field are written to the configuration file without quotes. In my case there was a # in the password and so login was failing. I worked around that by putting quotes <I>in the password field</i> but of course that's wrong - if there's a quotation mark in the password it should be interpreted as a quotation mark.

Jackysi, 2017-04-25:
This issue is unrelated to the AdvancedTomato GUI Modification. This whole project is only GUI modification meaning I don't change any internal software/code related to the routing it self. This project is goaling to stay on same state as Shibby's Tomato and for that reason this report is completely useless here. https://bitbucket.org/pl_shibby/tomato-arm/issues?status=new&status=open

closed 2017-04-25 by Jackysi

alanruttenberg, 2017-05-08:
I will submit an issue there. Thanks for the pointer.