Jackysi/advancedtomato #312: Possible bug in pppd?

PromoFaux, 2017-04-23:
I can't be entirely certain, as I only have a limited knowledge of all this,  but if I select "Manual" in the DNS settings under basic network settings, then those are the only DNS servers that should be used, right?

[image]

I have another modem/router in bridge mode, and am using AdvancedTomato (latest version) on a Netgear R7000. What I've noticed is that `/tmp/ppp/wan_options` always has the option `usepeerdns` set, no matter what I select on the settings page. This is causing `/etc/resolv.conf` or `/etc/resolv.dnsmasq` to also include my ISPs DNS servers, which ideally I would rather not happen. Edits to `/tmp/ppp/wan_options` get overwritten when `pppd` is restarted. Am I doing something wrong?

Jackysi, 2017-04-25:
This issue is unrelated to the AdvancedTomato GUI Modification. This whole project is only GUI modification meaning I don't change any internal software/code related to the routing it self. This project is goaling to stay on same state as Shibby's Tomato and for that reason this report is completely useless here. https://bitbucket.org/pl_shibby/tomato-arm/issues?status=new&status=open

closed 2017-04-25 by Jackysi

PromoFaux, 2017-04-25:
Apologies!