Jackysi/advancedtomato #317: Invalid IP address and Invalid number. Valid range: -1-32767

Ivantosov, 2017-05-08:
On save OpenVPN client settings displays: "Invalid IP address", and "Invalid number. Valid range: -1-32767".
On save basic-network settings displays: "The field "wan_modem_ipaddr" is invalid. Please report this problem."
Asus RT-N16, Tomato Firmware 1.28.0000 MIPSR2-3.4-138 K26 USB AIO
Router configuration backup:
[tomato_v128_mE8CBC0.zip](https://github.com/Jackysi/advancedtomato/files/983209/tomato_v128_mE8CBC0.zip)

[image]
[image]
[image]

Jackysi, 2017-05-08:
What sort of reset? I'm gonna look into it but i'm not sure i can fix it. Does this issue exist on Shibby's firmware too?

Ivantosov, 2017-05-08:
I can't currently turn off the router and change the firmware, but i think this is a GUI bug. It contains and saves a lot of hidden fields, which incorrectly filled after page refresh. At the moment I have to reset the router settings completely to get these errors disappear. Config partially reset buttons would help in this situation...

Jackysi, 2017-05-08:
That's really not a solution, even if it were NVRAM defaults only work for whole NVRAM. Best solution is to simply fix the issue that appears in the GUI :). I will look into it but not in this update. 

People are anxious and are annoying me daily when will I release 140 update :)

tomatolover69, 2018-01-06:
Dear Jackysi,

unfortunately I face the same issue as Ivantosov regarding the OpenVPN client settings, displaying
"Invalid IP address", and "Invalid number. Valid range: -1-32767"."

I really appreciate any tip or solution.

Best regards,
tomatolover

kollaesch, 2018-03-01:
Hi,
I got the same problem here yesterday. But I suspect something different.
I checked the available nvram-space. It was 0,01%. So I suspect that the 
my changes of the config (added openvpn-keys) config was getting to big.

So the script warns with 'invalid IP address' that it apparently can read/access 
a specific IP address in the nvram-settings any more, since this was not written
due to no available space.

So I should check for available space, I think.

jmoverley, 2018-03-30:
Found out default parameter on openvpn client2 settings -> advanced -> connection retry was blank and required a number! wouldn't save until set, this was discovered by checking the js code in using chrome debug console and looking at the vpn_client.asp

PTZ-M, 2018-09-28:
1.28.0000 MIPSR2-3.5-140 K26 USB AIO- similar problem, and Shibby's firmware too

peterjw, 2018-10-26:
> Found out default parameter on openvpn client2 settings -> advanced -> connection retry was blank and required a number! wouldn't save until set, this was discovered by checking the js code in using chrome debug console and looking at the vpn_client.asp root cause : bad default params in UI when no NVRAM data available

To get rid of the "Invalid IP address" part, I had to choose "Static Key" for _Authorization Mode_ from the _Basic_ tab and enter some dummy IPs here.

martin-mmmm, 2019-01-30:
Hey, still have the same problem, full nvram/invalid ip. is there any solution yet?
Tomato Firmware 1.28.0005 124 ND VPN

tomatofan222, 2020-04-05:
I know this is an old thread, but for those who Googled this issue and ended up here, there's a workaround for the issue of running out of NVRAM space due to the size of TLS certificates for OpenVPN that was successful for me. It involves uploading the certificates to JFFS, and referencing those files in your OpenVPN config. Here are the basic steps:

1. Enable JFFS. I mounted mine at /jffs
2. Copy the necessary certificates files to the tomato wherever you mounted JFFS (I used SCP to copy them, but there are other methods).
3. Adjust the authentication for OpenVPN to 'custom'
4. Put the parameters for authentication in the custom / options box. Here's a sample of the options:

```
dev tun
persist-tun
persist-key
cipher AES-256-CBC
auth SHA256
tls-client
client
resolv-retry infinite
remote XXX.XXX.XXX.XXX 1194 udp
verify-x509-name "OPEN.CERTIFICATE.COMMON.NAME" name
pkcs12 /jffs/CERTIFICATE.FILENAME.p12
tls-auth /jffs/TLS-KEY.key 1
remote-cert-tls server
comp-lzo adaptive
```