Jackysi/advancedtomato #318: Invisible WAN settings block

Ivantosov, 2017-05-08:
If disable WAN - wan block becomes invisible. So there's no way to turn it on again.
Asus RT-N16, Tomato Firmware 1.28.0000 MIPSR2-3.4-138 K26 USB AIO
Router configuration backup:
[tomato_v128_mC2EAB4.zip](https://github.com/Jackysi/advancedtomato/files/983247/tomato_v128_mC2EAB4.zip)
[image]

Jackysi, 2017-05-08:
Good catch, Will look into it. 
I suggest you stop sharing your configuration, could be dangerous!

Ivantosov, 2017-05-08:
I know, but my configs does not contain any information, that could lead to the vulnerability of my network...