Jackysi/advancedtomato #323: Empty "Overview"

falcon4fun, 2017-05-14:
status-data.jsx answers empty response on http. ;/

[image]
[image]

Jackysi, 2017-05-14:
Pretty much useless without router model and build. I hardly doubt this is AdvancedTomato bug since GUI files are not "dynamically" loaded. So the HTTPD handler it self is not returning file.

falcon4fun, 2017-05-14:
@Jackysi first screenshot is https and https works correctly. Second is http and response headers is missing, so response is empty too. But can't reproduce again that bug ;/
Router model: r6400.

Hm. Something is related to "Save" settings and service restarting I think.
Tryed clicking "save" and going from down menus to up.
https://i.imgur.com/9KGTvjf.png

Last logs was related to:
> May 14 22:43:43 Router daemon.notice miniupnpd[12542]: version 2.0 started
> May 14 22:43:43 Router daemon.notice miniupnpd[12542]: HTTP listening on port 2546
> May 14 22:43:43 Router daemon.notice miniupnpd[12542]: Listening for NAT-PMP/PCP traffic on port 5351
> May 14 22:43:43 Router daemon.info httpd[12544]: Tomato interface started successfully

I don't know how I can help more. Too random steps.

Hm. And again.. Log:

> May 14 22:43:50 Router daemon.debug dnscrypt-proxy[12084]: resolver timeout (UDP)
> May 14 22:43:56 Router daemon.info dnsmasq-dhcp[12051]: RTR-ADVERT(br0) 2001:xxx:xxxx::
> May 14 22:44:12 Router daemon.info dnsmasq-dhcp[12051]: RTR-ADVERT(br0) 2001:xxx:xxxx::
> May 14 22:44:26 Router daemon.info dnsmasq-dhcp[12051]: RTR-ADVERT(br0) 2001:xxx:xxxx::
> May 14 22:44:38 Router daemon.info dnsmasq-dhcp[12051]: RTR-ADVERT(br0) 2001:xxx:xxxx::

Killed httpd service from telnet and started again - "Overview" look normal.

Jackysi, 2017-05-14:
These are unrelated to HTTP, well I guess its likely its something connected with network devices / httpd app. Either way I don't work on HTTP software because it requires C coding knowledge which I don't have :(. So you will have to take this issue to Shibby who does have that knowledge =)

closed 2017-05-14 by Jackysi