Jackysi/advancedtomato #324: 2.4GHz WiFi disapears

TheBauwssss, 2017-05-17:
Clean installation of Version 3.4-140 for Linksys EA6700 (I installed EA6700-AT-ARM-3.4-140-AIO-64K.trx).

After configuration my 2.4GHz WiFi is gone from the landing page and it is no longer broadcast and available to other devices. The 5 GHz sill works fine.

[image]

Jackysi, 2017-08-03:
Strange, did you clear NVRAM when flashing?

closed 2017-08-20 by Jackysi

TheBauwssss, 2017-08-20:
Yes, I have reset the NVRAM immediately after flashing.

I have had to re-flash AdvancedTomato yesterday because the router had reverted to the original firmware after a power failure. 

The same problem appeared. Immediately after the flash both wireless interfaces were available, but after the reboot (because I changed a few QoS settings) the 2.4 GHz interface had disappeared.

I flashed the original Linksys firmware 3 times before flashing advanced tomato v1.38 and so far it has not reverted to the original firmware. The wireless interface has bugged out again though. NVRAM reset did not help.

I am not the only one having this problem:

http://www.linksysinfo.org/index.php?threads/e3200-with-shibby-latest-build-5ghz-disappear-after-reboot.69760/
https://www.myopenrouter.com/forum/shibby-tomato-v129-no-24ghz-wifi
https://www.reddit.com/r/HomeNetworking/comments/4xqv2a/24ghz_disappears_ea6500_with_advancedtomato/

I did find this thread: https://github.com/Jackysi/advancedtomato/issues/153

I'll go ahead and report it to the Shibby developers.

TheBauwssss, 2017-08-20:
I am as of yet unable to find a proper location to deliver this bug report to. Could you point me in the right direction?

I have found this: http://polarcloud.com/tomatofaq#how_do_i_send_a_bug_report

But that seems to be an old version and a different build.

Jackysi, 2017-08-20:
I'm not sure where Shibby would want bug report. He only uses issues tracker on Bitbucket at the moment I think. However its for ARM builds so it may not be correct way to report E3200 problems. Link is located on AdvancedTomato contact page here: https://advancedtomato.com/contact

q74cll, 2017-08-24:
Hey,
I got the same Problem on a Linksys EA6900 with Tomato 3.4-140 - after setting up 2.4GHz-Wifi and a Reboot the interface disappears. If you solve the problem or have a workaround please share.

tegila, 2018-12-27:
I'm on _Tomato Firmware 1.28.0000 -3.5-140 K26ARM USB AIO-64K_ *EA6900* having exactly the same problem. After reset the interface is up, after some reboots the 2.4Ghz disappear =\

Wonderful firmware did you guys have also.
Thanks _/\_

JacobJacob, 2019-03-17:
does anybody has a solution for this issue ? ATT 3.5-140, EA6700, has the same issue.

TheBauwssss, 2019-03-17:
I still have the problem, ended up putting a UPS on the thing so it never had to reboot. Export the configuration, reflash the firmware and then reupload the saved configuration in case it does.

I replaced it with some new UniFi WiFi access points two months ago, they have been working great so far and I have had 0 issues with them.