Jackysi/advancedtomato #328: Lots of timeout errors with ASUS RT-N16 and 1.28.0000 MIPSR2-3.4-140 K26 USB AIO

dtech0501, 2017-05-30:
I have two RT-N16s that almost always show the attached error when switching from page to page. This did not happen with version 3.4-138. It also does not happen on my Netgear 6300v2 with 3.4-140. Firefox 53 on Linux Mint 18.1

[image]

dtech0501, 2017-05-30:
I'm getting these timeout errors (screenshot below) about every 30 seconds on the bandwidth page.

[image]

Jackysi, 2017-06-30:
Firmware issue. I can't help with this, please pass it to another firmware developer like Shibby.

closed 2017-06-30 by Jackysi