Jackysi/advancedtomato #329: OpenDNS DDNS updater does not work.

ryanneufeld, 2017-05-31:
The OpenDNS updater is currently making requests to http://opendns.com/nic/update?hostname=<hostname>&myip=@IP

It needs to be updated to http://**updates**.opendns.com/nic/update?hostname=<hostname>&myip=@IP

Even better however would be to update the plugin to use:
https://**updates**.opendns.com/nic/update?hostname=<hostname>&myip=@IP

ryanneufeld, 2017-05-31:
I just found this bit of code https://github.com/Jackysi/advancedtomato/blob/bb25e9fafc703dc678498b899f74aacb43191897/release/src/router/mdu/mdu.c#L1779

Which if i'm reading this correctly, should be using the right url, but it appears to be doing so over http?

Jackysi, 2017-06-30:
I'm sorry but this issue is unrelated to the UI and currently I can only work on UI part of Tomato. Please pass this issue to Shibby or another dev who works on Tomato Firmware.

closed 2017-06-30 by Jackysi

ryanneufeld, 2017-06-30:
No problem