Jackysi/advancedtomato #331: USB & NAS options not available

Davidhanson90, 2017-06-10:
Hey, 

I have advanced tomato on my netgear 7000.  
1.28.0000 -3.4-140 K26ARM USB AIO-64K

I wondered if USB and NAS support is available? I cannot seem to see in in the gui but the firmware supports it. A sniff around the JS code seems to suggest some support

```
function navi_icons(b) {
    switch (b) {
    case "Status":
        return "home";
        break;
    case "Basic Settings":
        return "hammer";
        break;
    case "Advanced Settings":
        return "shield";
        break;
    case "Port Forwarding":
        return "forward";
        break;
    case "Quality of Service":
        return "gauge";
        break;
    **case "USB & NAS":
        return "drive";
        break;**
    case "Web Services":
        return "cloud";
        break;
    case "VPN":
        return "globe";
        break;
    case "Administration":
        return "wrench";
        break;
    default:
        return "plus";
        break
    }
}
```

 but the nav options structure doesn't have it so think its unavailable currently. 

```
    var d = {
        Status: {
            Overview: "status-home.asp",
            "Device List": "status-devices.asp",
            "Web Usage": "status-webmon.asp",
            Logs: "status-log.asp"
        },
        "Basic Settings": {
            Network: "basic-network.asp",
            IPv6: "basic-ipv6.asp",
            Identification: "basic-ident.asp",
            Time: "basic-time.asp",
            DDNS: "basic-ddns.asp",
            "DHCP/ARP/BW": "basic-static.asp",
            "Wireless Filter": "basic-wfilter.asp"
        },
        "Advanced Settings": {
            "Access Restriction": "advanced-restrict.asp",
            "Conntrack/Netfilter": "advanced-ctnf.asp",
            "DHCP/DNS": "advanced-dhcpdns.asp",
            Firewall: "advanced-firewall.asp",
            Adblock: "advanced-adblock.asp",
            "MAC Address": "advanced-mac.asp",
            Miscellaneous: "advanced-misc.asp",
            Routing: "advanced-routing.asp",
            "MultiWAN Routing": "advanced-pbr.asp",
            Wireless: "advanced-wireless.asp",
            VLAN: "advanced-vlan.asp",
            "LAN Access": "advanced-access.asp",
            "Virtual Wireless": "advanced-wlanvifs.asp"
        },
        "Port Forwarding": {
            Basic: "forward-basic.asp",
            "Basic IPv6": "forward-basic-ipv6.asp",
            DMZ: "forward-dmz.asp",
            Triggered: "forward-triggered.asp",
            "UPnP/NAT-PMP": "forward-upnp.asp"
        },
        "Quality of Service": {
            "Basic Settings": "qos-settings.asp",
            Classification: "qos-classify.asp",
            "View Graphs": "qos-graphs.asp",
            "View Details": "qos-detailed.asp",
            "Transfer Rates": "qos-ctrate.asp",
            "B/W Limiter": "qos-qoslimit.asp"
        },
        VPN: {
            "OpenVPN Server": "vpn-server.asp",
            "OpenVPN Client": "vpn-client.asp"
        },
        Administration: {
            "Admin Access": "admin-access.asp",
            TomatoAnon: "admin-tomatoanon.asp",
            "Bandwidth Monitoring": "admin-bwm.asp",
            "IP Traffic Monitoring": "admin-iptraffic.asp",
            "Buttons/LED": "admin-buttons.asp",
            "CIFS Client": "admin-cifs.asp",
            Configuration: "admin-config.asp",
            Debugging: "admin-debug.asp",
            JFFS: "admin-jffs2.asp",
            Logging: "admin-log.asp",
            Scheduler: "admin-sched.asp",
            Scripts: "admin-scripts.asp",
            Upgrade: "admin-upgrade.asp"
        }
```

Is this something due to be supported or do I have an issue with my setup?

Jackysi, 2017-06-10:
Well, the USB & NAS should be visible on any R7000. That's weird. Which image did you flash?

Davidhanson90, 2017-06-14:
Just to let you know this fixed itself so could be transient. Can't provide much more info other than I had scheduled reboot setup

closed 2017-06-14 by Davidhanson90

shanDB, 2017-08-24:
I had the same issue, same router, same firmware as the OP.
I was also using Chrome as my browser, which had cached an older version of the admin page. ONce I did a hard refresh the USB options appeared in the Admin menu.

Fury180, 2018-08-05:
Same Problem here! I have an R7000 I just flashed Advanced Tomato on for the first time with version "v3.5-140AIR-64K"

Jackysi, 2018-08-05:
Well I can't influence Chrome's cache. So its best to reload cache on flash. Same goes for NVRAM.

Mattwmaster58, 2018-12-22:
I also encountered this issue, I tried a cache clear but it didn't work. My model is the R7000 and I've flashed `Tomato Firmware 1.28.0000 -3.5-140 K26ARM initial-64K`

Mattwmaster58, 2018-12-22:
@Jackysi Can't you plunk these meta tags into the head tag?
```
<meta http-equiv="Cache-Control" content="no-cache, no-store, must-revalidate"/>
<meta http-equiv="Pragma" content="no-cache"/>
<meta http-equiv="Expires" content="0"/>
```