Jackysi/advancedtomato #332: Cannot set vlans properly

DBLouis, 2017-06-16:
Netgear WNR3500Lv2
AdvancedTomato 3.4-140

Vlans settings are not saved correctly. I set it like this:
vid 0: ports 8, bridge br1, default
vid 1: ports 1 2 3 4 8(tagged), bridge br0
vid 15: bridge wan

And after reboot it is modified:
vid 0: ports 8, bridge br1, default
vid 1: ports 1 2 3 4, bridge br0
vid 2: ports 8
vid 15: bridge wan

(port 8 is wan, bridged to br0)

DBLouis, 2017-06-16:
I tried other configurations, every times it add or remove a vid, change the default vlan, swap bridges ...

wbinek, 2017-07-09:
I've also noticed the issue. I tested vlans with original tomato shibby 140 and it works fine there.

DBLouis, 2017-07-14:
Someone has to do something about it, my router is basically unusable. I tried using the CLI, it is the same. It keeps adding another vlan and messing with the ports ...

Jackysi, 2017-07-14:
Maybe this will fix the problem: https://github.com/Jackysi/advancedtomato-gui/commit/d77af3574dddc1a1e742f22e3ee0eb2fbb622b3a

Jackysi, 2017-08-03:
Duplicate: https://github.com/Jackysi/advancedtomato/issues/338

closed 2017-08-03 by Jackysi