Jackysi/advancedtomato #337: bricked

blaze2051, 2017-07-23:
i was on a stock netgear wndr 3700v3 and installed the latest firmware. router never reboot, its bricked.
i have access to the router via ttl putty serial port, i need details on how to force flash a stock firmware or a working advanced tomato firmware.

Jackysi, 2017-07-23:
Did you try clearing NVRAM before flashing the router?

blaze2051, 2017-07-23:
Honestly no, I never have in the past and I'm not sure if that would have helped. From what I can read in the CFE loading of file failed file too large

blaze2051, 2017-07-23:
I managed to install the latest stock firmware from netgear and it up and running once more. I when ahead and flashed a mini ddwrt and its working. I just want to know why the tomato caused a semi bricked cause the stock recovery was not functional so serial was the only way

blaze2051, 2017-07-23:
Aslo what isn't there a detail guide on how to flash properly from stock to advanced tomato?

Jackysi, 2017-08-03:
There is no guide, no. I don't write guides, I let people do that and if they send me links to guides I add them to the web site. I wrote that on Linksysinfo forums way back and I just stopped bothering because no one helped / turned up. When flashing routers you always have to clean NVRAM. Always. 

I guess this router is not supported then. See: http://anon.groov.pl/index.php?model=Netgear+WNDR3700v3

I will remove it one day soon.

blaze2051, 2017-08-03:
The link you provided has it listed but cant find link to get firmware

trismos, 2017-09-24:
Can I leave a suggestion here? a well, I am a truck driver on good meds, gonna anyways... If your gonna dick around with stuff you know nothing about, consider it "bricked" before doing so.... just my opinion... I would have read up on what needs to be done, have it all in order, then what you need in case it fails... mortor/grout, matching colors... 
 I have an Asus RT-N53 A1 (the A1 version is the worse of them all, something Asus put out on the market to screw with everyone) anyways, meds are starting to wear off, so, short story: 
 tomato works great, even got it on with an older version, first try, did a recovery though... for Asus, its reboot while holding reset button, but hey, probably wont work for you.. why? dont know, I know it worked for mine, cause I did some reading first... 
 but the thing to do for clearing NVRAM? well, you could try the basics? whats that? I dunno.. do some reading? try here: https://www.dd-wrt.com/wiki/index.php/Hard_reset_or_30/30/30
 but what do I know, I am just a truck driver... must be the good meds... 
 anyways, back to me, I put MY router into recovery... probably bricked it first.. no, no, we are past that part of the story, anyways, did the recovery thing, has options to do other things, but I figured what the hell, just load the earlier version of Tomato, and BAM!! I got a brick, no wait, did the recoverything again, and tomato had a baby girl... then upgraded to this bad boy... works like a charm on the piece of crap Asus... 
... hey, this is about me now... Jacky just makes the bells and whistles... cant help it if its not playing your tune... oh, sorry, didnt know this was the Netgear help page, sorry, ignore everything above...

sitekeeper, 2017-09-24:
There are guides and good videos on youtube, Google is your friend!
Oh wait look http://tomato.groov.pl/?p=379
BE WARNED! This is highly beta tomato version. This image may brick your router. Make sure you have serial cable to debrick router!!

Google a 30 30 30 reset, I would try a 10 -10 - 10

closed 2017-11-14 by Jackysi