Jackysi/advancedtomato #338: Using tagged VLAN for WAN port

j-vizcaino, 2017-07-24:
First, thank you for putting AdvancedTomato together: this a really good piece of software that gets the job done and has a nice and pleasant web UI.

### Environment

* Netgear R7000
* Tomato Firmware 1.28.0000 -3.4-140 K26ARM USB AIO-64K

### What I am trying to do

* Use untagged vlan1 for LAN `br0` (4 switch ports + 2 wifi interfaces)
* Use tagged vlan with VID=1 for eth0 port
* Use tagged vlan with VID=10 for eth0 port
* Vlan 10 is meant for WAN

### What I tried to do

* vlan2 -> set VID to 10
* check the tagged checbox for `WAN port`
* vlan1 -> set VID to 1
* check tagged for `WAN port`, check the 4 ports and the default option
* apply changes, the router reboots

When reboot is done and I go to the Avanced/VLAN page, I cannot see the tagged column checked for the LAN or the WAN bridge line (I suspect a bug in the UI).
Also, the WAN port that is supposed to receive an IP from the DSL box never gets that (it should receive DHCP response on vlan10)

Logging into the router, I can see 

```
$ ip l
...
9: vlan1@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP
10: vlan10@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP
```
If I try to set a static IP to `vlan10` I still cannot ping the DSL box.

Am I doing something wrong or is there a bug in the firmware preventing me from getting this architecture up and running?
Any hint or help would be greatly appreciated.
Thank you.

Jackysi, 2017-08-03:
I believe the Advanced VLAN pages have some sort of bugs, but I'm not sure. TVLZ did fix some stuff few weeks ago, but there is no update on firmware yet. Shibby didn't release any updates and I'm pretty busy too :(. I haven't really had the time to work on AT last few months.

You may try compiling firmware on your own, or download web files, store them on USB/HDD and run web interface through custom "path". This is very advanced stuff, but its the only fix at the moment.

j-vizcaino, 2017-08-03:
@Jackysi Thank you for your answer. Running short on time as well so may not be able to try the compile + run dev build soon. I had another idea about how my setup could be wrong so I will dig into this and post updates here.