Jackysi/advancedtomato #339: Shibby v140 needs tagging

afontenot, 2017-07-31:
The latest downloads are based on Shibby v140, but that release isn't tagged in this repository. Can you tag it so we know what revision the downloads are based on?

Also, since (I think) this project is GPL licensed, it would probably be good to include it in the root of the source dir. 

Thanks!

Jackysi, 2017-07-31:
I'll add tag. About GPL I'm not sure what you mean, No other tomato author has any info about GPL.

Jackysi, 2017-08-03:
added tags, but these should be done by Shibby, so in future I won't be adding any tags unless he does.

closed 2017-08-03 by Jackysi

afontenot, 2017-08-04:
Thanks for the tag. With respect to the GPL, I was assuming because of the FAQ "AdvancedTomato is based on many open source projects which are licensed under GPL license", that AdvancedTomato constitutes a "derived work" under the GPL, and therefore must be GPL licensed as well. Many authors of GPL software put a `LICENSE` or `License.txt`file containing the GPL so that it's obvious how the software is licensed.