Jackysi/advancedtomato #343: Add stealth mode configuration in WebGUI

Olangu, 2017-08-20:
I was expecting to find configuration for turning off and/or dimming the LEDs under Administration -> Buttons/LED, but no such configuration was available so instead I had to enable SSH and do the stealth configuration from there.

Please add LED control into the WebGUI.

Jackysi, 2017-08-20:
I don't extend router functionality in that way. You should take that to Shibby or Kille72

closed 2017-08-20 by Jackysi

Olangu, 2017-08-20:
I don't follow, the functionality is in the software using the command stealthMode.

```
root@tomato:/tmp/home/root# stealthMode 
stealthMode Sunset v0.10 by Monter
Usage: stealthMode {on|off|sun <city_code>|sun_off|sch_on <H> <M>|sch_off <H> <M>|sch_clear|clear_all}

 [Standard mode]
   on | off        - enable or disable steathMode in real time

 [Sunset mode]
   sun <city_code> - will daily get the sunrise and sunset times of a specific
                     location automatically and activate Sunset mode
                     To be able to determine your <city_code> you need to first
                     go to http://weather.yahoo.com/ and look up your location
                     The last NUMBERS in the URL will be your <city_code>
                     Example: "stealthMode sun 514048" if you live in Poznan/Poland ;)
                     This feature requires a working Internet connection
   sun_off         - fully deactivate Sunset mode

 [Scheduled mode]
   sch_on <H> <M>  - set the hour and minutes of the scheduled enable/disable
   sch_off <H> <M>   stealthMode in Standard mode and adding jobs to the Crontab
                     Hour and minute time must be a numbers without any additional
                     characters, where hour is a mandatory parameter, while not
                     specifying an minute will assign a default 00 value
                     These options add just the right job for Crontab, nothing more
   sch_clear       - removes tasks from Crontab for scheduled enable/disable
                     stealthMode function set by sch_on and sch_off switches

 [Repair / debug]
   clear_all       - removes all jobs from Crontab and completely disables all
                     available stealthMode modes

```

Jackysi, 2017-08-20:
HTTPD handler needs changes for this to work. I stick to Shibby's. Also functionality vise I'm not adding features but keeping the same as Shibby has. For reasons like easier GIT management and many more. But I'm not really going to explain that in detail. 

in short, I don't extend features. I only modified and redesigned GUI. I never ever said I will add more features to Tomato. If Shibby decides to QUIT tomato project, than maybe things will change. but Maybe. I'm very very busy and I have no time to learn whole back end of the Tomato and Linux compilations. Also Tomato is C based project which is not my skill set.