Jackysi/advancedtomato #345: Couldn't update from 138 to 140 on Huawei WS880

s10, 2017-08-23:
After pressing "Upgrade" button, upload doesn't start (no data seemed to be sent), and then after a few seconds all connections to router stop (no ping to ethernet router ip, wireless is off).

Jackysi, 2017-11-14:
Um, so router bricked it self? I really don't know what that could be. Did you try clearing NVRAM?

s10, 2017-11-16:
No, i didn't try clearing NVRAM. I've been upgrading advancedtomato firmware since 2015-08 without clearing NVRAM on each upgrade, it worked well for me.
If I reboot router after several minutes (5-10 or more) since pressing "Upgrade" button (after connectivity loss), it just boots with old firmware and with old configs. But if I set "clear NVRAM", i'm not sure, that router will not be bricked.

AlexeyMoiseev, 2018-02-06:
Same problem.

elect0nik, 2018-03-25:
Same problem

closed 2023-01-13 by s10