Jackysi/advancedtomato #346: Guest Wi-Fi Internet Access

WillyTP, 2017-08-25:
Hello everybody.
I need some advice regarding Wi-Fi Internet access for guests.

Actually I don't understand properly if this is a GUI bug (I mean, that the GUI doesn't apply some policy rules when settings), if it's strictly related to Tomato or if it's supposed to work like this.

Anyway, to make it simple: I've setup a Wi-Fi guest network like described [HERE](https://learntomato.com/setup-guest-network-guest-wifi-tomato-vlan/).
This way, guest Wi-Fi works but I'm unable to access to the Internet.

Following some advices around, by adding the following code to Administration ->Scripts -> Firewall Tab

```
iptables -t nat -I POSTROUTING -o br0 -j SNAT --to $(nvram get lan_ipaddr)
iptables -I FORWARD -i br1 -m state --state NEW -j ACCEPT
iptables -I FORWARD -i br1 -m iprange --dst-range 192.168.1.2-192.168.1.250 -j REJECT
iptables -I INPUT -i br0 -p tcp --dport 67:68 --sport 67:68 -j REJECT
iptables -I INPUT -i br0 -p udp --dport 67:68 --sport 67:68 -j REJECT
```

Eveything works fine.
Is it supposed to work just like this, or am I missing some settings, is it a bug or what?
Thanks for information!

WillyTP, 2017-08-28:
I'm sorry, I forgot to mention that Guest Wi-Fi Internet Access didn't work under a special circumstance, using the router as second router connected in cascade with another one which doesn't support VLANs.
Please consider this as bug as INVALID.

Thanks

closed 2017-08-28 by WillyTP

Jackysi, 2017-08-29:
Its okay :)