Jackysi/advancedtomato #347: R8500? :D

Extarys, 2017-08-28:
Would the R8000 firmware work on the R8500? 

I just looked at the spec and they are similar - but I do know the risk of flashing a firmware from another device.

R8500 tomato doesn't give me any results and I would love to give it a try sometimes. I'm sick of the ugly ddwrt interface :P

Thanks!

rat-s-tar, 2017-09-03:
If you have a look over at shibby's project page issue tracker. A lot of R-series routers are getting bootloops when WAN cable is plugged in. I'd avoid until this is fixed as I tried it and while it was a great fw, the reboots and loops were a dealbreaker.

https://bitbucket.org/pl_shibby/tomato-arm/issues?status=new&status=open

Jackysi, 2017-11-14:
AdvancedTomato is based on Tomato by Shibby. I don't do any additional or extra firmware support because I don't have knowledge to do so. So I suggest you wait for Shibby and once he supports it and its safe, I will add the router as well.

closed 2017-11-14 by Jackysi