Jackysi/advancedtomato #356: Support for RT-N12B1/C1/D1

waqaslam, 2017-10-05:
Hi
As per "Tomato by Shibby" description, his firmware is compatible with RT-N12B1/C1/D1 through K26RT-N.
Since the base code of your firmware comes from shibby, so would it be safe to assume that your firmware for RT-N10U can also be installed on RT-N12B1/C1/D1?

Here's the shibby's list of routers K26RT-N can support - which also includes RT-N10U and RT-N12B1/C1/D1.

> K26RT-N – MIPSR2 – special builds for E4200, RT-N10U, RT-N12B1/C1/D1, RT-N15U, RT-N53, RT-N66U, WNR3500Lv2 and newer Linksys E-series routers.

regards

Jackysi, 2017-10-05:
AdvancedTomato is fork with new GUI. I don't really know on which routers it works other than confirmed by other people. Biggest issue is Flash size of the router, it needs to be above 8mb in most cases.

closed 2017-10-05 by Jackysi

waqaslam, 2017-10-05:
RT-N12B1/C1/D1 has the exact same flash/ram as RT-N10U. Though the cpu chipset they use are slightly different (N10 -> BCM5356U AND N12 -> BCM5357C0).
Since you do support N10, so I guess it should be fine to install that firmware on N12 too.

waqaslam, 2017-10-06:
I hereby confirm that it's possible to flash AdvancedTomato on RT-N12B1/C1/D1.
I tested it by installing firmware under RT-N10U on my **RT-N12D1** router and it worked perfectly fine.
If possible, please update your supported-devices page to contain this model too.

[image]