Jackysi/advancedtomato #364: Key Reinstallation Attacks on WPA2

carofe82, 2017-10-16:
Is Advanced tomato affected by Krack?
https://www.krackattacks.com/

carofe82, 2017-10-16:
I see that it is being tracked here: https://bitbucket.org/pl_shibby/tomato-arm/issues/184/apparently-wpa2-protocol-is-vulnerable-to.

chrismetcalf, 2017-10-17:
So are we waiting for a fix to base Tomato first? What's the process by which those updates make it into AdvancedTomato?

sitekeeper, 2017-10-17:
I am not sure why everyone finds this so hard to understand, AdvancedTomato is only a Graphics Interface enhancement!!!!!!! All changes for the Firmware has to come from Shibby. 

As for as this KRACK goes you are more likely to be hit by lighting on 2 consecutive Tuesdays in one month then be affected by this.

Jackysi, 2017-10-17:
> I am not sure why everyone finds this so hard to understand, AdvancedTomato is only a Graphics Interface enhancement!!!!!!! All changes for the Firmware has to come from Shibby.
>
> As for as this [...]
Thanks. I got about 40+ mails about this topic. Not to mention issues tracker. AdvancedTomato web site should explain that this is GUI but well people can't read in 2017. This problem is also part of clients and isn't 100% router related.

chrismetcalf, 2017-10-17:
Yup, I totally get that this is an issue in base Tomato and not AdvancedTomato. I inquired to ask what the process will be for an updated version of AdvancedTomato once fixes are ready upstream.

Keeping an issue open and being transparent about the process will hopefully serve you a lot better.

BTW, I love AdvancedTomato and find it far superior to base Tomato. Thanks for all your hard work!

Jackysi, 2017-10-21:
Here is the discussion (its going to take off): http://www.linksysinfo.org/index.php?threads/krack-vulnerability-in-tomatousb.73823/