Jackysi/advancedtomato #371: Parsing error of ampersand symbol in RADIUS shared secret

maurice-w, 2017-11-16:
In _Advanced Settings / Virtual Wireless_, configure _WPA2 Enterprise_ security for one of the _Virtual Wireless Interfaces_ and enter a _Shared Key_ containg an ampersand symbol (&). When saving, the string gets truncated at the ampersand symbol.

I didn't test whether this also happens for other input fields in the GUI, but I wouldn't be surprised.

This is a new issue in 3.5-140; it worked fine in 3.4-140!

Router: Linksys E4200

Jackysi, 2017-11-16:
That sounds the opposite of what we did in last release. Can you show example?https://github.com/Jackysi/advancedtomato-gui/pull/31

If that's the case this is a big problem.

maurice-w, 2017-11-17:
I agree that this IS a big problem. Here are two more things I tried:
1. Navigate to _Basic Settings / Network_.
2. Enable _WPA2 Personal_ security with AES encryption, enter the shared key `12345&67890`, click _Save_.
3. An error message will be displayed: _The field "wl_wpa_psk" is invalid. Please report this problem._
4. This is bad because, by definition, a WPA2-PSK may contain any printable ASCII character.
5. Now switch security to _WPA2 Enterprise_, enter the same shared key and click _Save_ again.
6. No error message is displayed, it looks like everything is OK.
7. Navigate to another page, then go back to _Basic Settings / Network_ and click into the _Shared Key_ field.
8. Only the string `12345` is displayed, `&67890` has been truncated.
9. This is even worse because the GUI saves a modified value without displaying an error message.

Jackysi, 2017-11-17:
I will revert this commit and re-compile images. I am not sure why I merged this, I was sure it will bring issues... Some one just liked to have their unicode characters on wifi names lolz and they changed global function that messed up every thing. And I was stupid enough to merge it.

closed 2017-11-17 by Jackysi

noplanman, 2018-03-25:
@Jackysi The ampersand and semicolon is also an issue when saving the custom script in `Administration -> Buttons/LED` for the Netgear R7000. It isn't correctly url encoded, so everything after the first `&` or `;` (and possibly other characters too) is lost.