Jackysi/advancedtomato #382: Init and Shutdown event scripts broken on 3.5-140

dgaglioni, 2017-12-15:
Hi Jackysi!

First of all thank you for the incredible work on AT GUI, I noticed that after a clean install (stock to 3.5-140) on my EA6700 that INIT and Shutdown scripts stooped working. 

**On 3.5-140 I could only see part of the script on NVRAM** 
"script_init=for line in `nvram show | grep =$ `" 
**instead of showing it full** 
"script_init=for line in `nvram show | grep =$ `; do var=${line%*=}; nvram unset $var;".

I'd like to mention that EA6700 and EA6500v2 (AFAIK) have a NVRAM/CFE BUG that does not allow to store more than 32KB of configs, so to apply this configuration first I apply the script below (to clean the NVRAM null values) on linux CLI and then apply on GUI to save it on NVRAM config.

Resuming I had to downgrade back to 3.4-140 and my scripts worked again :)

Scripts I use:

for line in `nvram show | grep =$ `; do var=${line%*=}; nvram unset $var;
done
nvram commit 
#

for line in `nvram show | egrep -o 'wan[3-4][^=]+'`; do var=${line%*=}; nvram unset $var;
done
nvram commit
#

sovanyio, 2017-12-25:
It looks like the web gui is truncating the scripts at semi-colons
I reformatted my scripts without them and saving scripts worked.

Jackysi, 2017-12-30:
The downgrade part sounds very strange since I have not changed any code that would affect this. Its possible that another function was shared with Scripts so maybe downloading image again and flashing it may fix the issue. Otherwise try as @sovanyio suggested. I have a lot of scripts and i have no issues as you mention above

chrisgh1, 2018-01-03:
I just wanted to say that I also am experiencing something similar when trying to write a script in USB & NAS -> USB Support - Hotplug script.  see conversation http://www.linksysinfo.org/index.php?threads/semicolon-and-scripts.73962/#post-293301 for what I've done so far.

sovanyio, 2018-01-03:
Hm actually @Jackysi this might be the PR I raised to remove the deprecated escape call to enable emoji. I could try to come up with something more robust but probably better to revert that change for now.

GLawlor-, 2018-01-15:
As a workaround you can set the nvram variables in SSH.  It will look correct on the Event Scripts after you refresh the page.
If you want to see the current settings then type:
 `nvram show | grep script_`

in SSH session:
`nvram set script_init='for line in nvram show | grep =$; do var=${line%*=}; nvram unset $var;done;nvram commit'`
`nvram set script_shut='for line in nvram show | grep =$; do var=${line%*=}; nvram unset $var;done;nvram commit'`

`nvram commit`

sovanyio, 2018-01-15:
@Jackysi looks like commit is causing all of the problems (sorry!). I just updated my branches with a fix that should retain the unicode functionality for SSID but revert the escape behavior everywhere else. https://github.com/sovanyio/advancedtomato-gui

Haven't built it yet since I lost my build vm but working on that now. Wanted to push that for feedback.

Jackysi, 2018-02-15:
Its okay =) we fixed it but requires re-build. I will get it done once I have enough changes

closed 2018-02-15 by Jackysi