Jackysi/advancedtomato #384: Wifi Password gets corrupted after saving

TheHellSite, 2017-12-28:
On the current AdvancedTomato build (3.5-140) for my asus rt-n18u I can't write my usual wifi password.
The following text is not my password, but it will show the bug just fine.

%WIFI$12§34$WIFI% ---> The field "wl_wpa_psk" is invalid. Please report this problem.
This password gives me the above text next to the save button and doesn't even save.

$Fr3e%1234%Fr3e$ ---> $Fr3e343e$
My password only contains $ and % as special characters.
And the % char gets replaced by this strange symbol.
To reproduce the error just type any password with % or § (maybe there are more characters that are bugged) and save it. Either reload the page or click on "Basic Settings --> Network". Then click into the shared key field. It should be corrupted.
[image]
[image]

Jackysi, 2017-12-30:
Can you re-download image and re-flash it? There was some differences after 2 days of release and I didn't want to put whole new release number because I only changed the way special characters are handled. Other than that I have no idea what could it be, I have no problems.

TheHellSite, 2017-12-30:
When did you release the latest image? I downloaded the currently flashed one on the 26th dec., I think.
But I can try downloading it again.

Jackysi, 2017-12-30:
About a month ago. So that's probably not the case. Strange though, I tried to do that and it works fine here.

TheHellSite, 2017-12-30:
I even cleared NVRAM after installing the update.

Jackysi, 2017-12-30:
That has no effect on the problem though.
Can u try setting password via Shell / Telnet / SSH?

Something like "nvram set wl0_wpa_psk=password" it really depends on your router ( I don't know exact values )

TheHellSite, 2017-12-30:
I will try! Doesn't work either! The password gets set, but extremely cut.
"nvram set wl0_wpa_psk=%Fre3$1234$Fre3%" will give me "e3".
Without special characters, it works fine.

Jackysi, 2017-12-30:
but did the login work via wifi? because otherwise its probably nvram limitation and can't be really fixed unless some one else does the fix. I develop gui only =)

TheHellSite, 2017-12-30:
So I just tried it with telnet.
"%Fre3$1234$Fre3%" gets to "%Fre3%".
I will try the 3.4-140 release now.

TheHellSite, 2017-12-30:
Same bug on 3.4 release. Very strange.
I even tried different browsers and devices to set the password.

user1969903, 2017-12-31:
The problem is that some of those special characters are used by the firmware. When you save the settings,  the router basically gets a string of text in the form of URL parameters such as 
`wl_wpa_psk=somePassword&someOtherSetting=someOtherValue&etc`
Notice that the "&" character is used to separate each setting. In this case the "&" is a special character and there are other such special characters that when you use in the password field, the router sees them as special characters.
To fix this, these characters need to be replaced with a code so that the router knows it's part of the password and not a special character. This is called encoding. For example, the "&" needs to be replaced with "%26". 

I'm going to open another issue and provide a temporary solution so that you can set a password as you want.

TheHellSite, 2017-12-31:
Okay cool, thanks for pointing that out!

TheHellSite, 2018-01-01:
@Jackysi 
I just tried the standard "1.28.0000 MIPSR2-140 K26 Max" build from shibby. No problems there.

Jackysi, 2018-02-15:
Fixed.

closed 2018-02-15 by Jackysi

mmartoccia, 2019-01-08:
Is this correction in a release yet?