Jackysi/advancedtomato #385: Wifi password is not correctly escaped

user1969903, 2017-12-31:
When setting the wifi password, for example the shared key in the case of WPA personal, the password does not get correctly escaped and is misinterpreted by the router.

The issue is at line 217 in the "tomato.js" file:
`v.push( escapeCGI( f.name ) + '=' + escapeCGI( f.value ) );`

The escapeCGI function just replaces the `+` character with `%2B`. A more appropriate solution would be to use the `encodeURIComponent `function that would correctly encode all special characters, at least for the password field.

As a temporary workaround, before clicking the save button, one can:

1. Open the browser developer tools (search online to find out how)
2. From the console tab, copy and paste the following (replacing the `your_password` with your actual password)
`var escapeCGI = function(s) {
    if (s == "your_password")
        return encodeURIComponent(s);
    else
        return s.replace(/\+/g, '%2B');
}`
3. Press enter. At this stage, you should see a message appear saying something like 'undefined'. If anything else appears, especially if it's red or has the word error, you've done something wrong.
4. Minimize the developer tools (don't close it, just minimize it or move it out of the way)
5. Click the save button

What this does is that it redefines the escapeCGI function so that it tests if the value the function is currently encoding is your password, it uses a proper URI encoding, if not, it does what the function did before changing it.

Jackysi, 2017-12-31:
The problem is that a lot other functions use this function. I still believe best solution would be fixing this within http handler (either encoding or what it accepts)

user1969903, 2017-12-31:
I'm not going to pretend I know the inner workings of the GUI or indeed a solution, I just thought I should point out where I think the issue is as well as a quick and dirty (non-permanent) fix for anyone like me who has a password that contains URI reserved characters.  
I installed the firmware yesterday for the first time and apart from this minor annoyance, it's working really great.

Jackysi, 2018-02-15:
Fixed here: https://github.com/Jackysi/advancedtomato-gui/commit/4f9fa7f22ba04bde5327e79b764c578bb6329fe9

closed 2018-02-15 by Jackysi