Jackysi/advancedtomato #386: 5GHz R7000 not working with any security (but works without security)

jacobfernando, 2018-01-02:
basically with any wpa or wep it does not connect even thou is broadcasting. but when i take off the security no password it does connect

TheHellSite, 2018-01-06:
You might wanna take a look at this and see if you are facing the same issue.
https://github.com/Jackysi/advancedtomato/issues/384

Also take a look at this workaround
https://github.com/Jackysi/advancedtomato/issues/385

jacobfernando, 2018-01-06:
But im not using any special character.

Jackysi, 2018-01-10:
I can not reproduce this. Please reset NVRAM and re-flash firmware.

jacobfernando, 2018-01-11:
I have reset the NVRAM and re-flashed, 5ghz broadcasting but device does not connect. If i remove security it does connect

Jackysi, 2018-01-11:
Faulty device?

jacobfernando, 2018-01-11:
Brand new

Jackysi, 2018-01-11:
I meant the one that doesn't want to connect to router =). I don't know how to help you, I have no issues with authorization nor any wifi modules. Got R7000 and AC3200 as primary. both run fine

jacobfernando, 2018-01-11:
Ok i will try again

Jackysi, 2018-02-15:
so?

malderhout, 2018-02-17:
Try to change the Country code of your 5Ghz Wifi (Advanced Wifi Settings) to "MALAYSIA". I had the same problem with the R7000. The 5Ghz Wifi works well now on the R7000.

edulmartins, 2018-08-08:
Hi. I just flashed my R7000 with tomato and I'm having this problem with my 5G network. I tried change to "Malaysia" and didn't work. Does anyone another idea?

malderhout, 2018-08-09:
You van also try to switch off choosing the channels automatic. Choose a manual lower channel for example channel 44. In my case not all the devices will work on a higher 5 GHZ channels

JLJTGR, 2018-12-26:
A little late to the thread, but what worked for me was changing the 5GHz MAC address away from the "default" to a more sane address.

[image]

For some reason the "default" had the MAC address start with 7A unlike the others.  When like this, no-security WiFi would work but otherwise it would not.  Manually changing 7A to 78 like my other MACs resolved the issue.  I have no idea where the 7A came from except that pressing the Default button makes it return.

spinko101, 2021-02-14:
@JLJTGR

thanks, that worked for me !