Jackysi/advancedtomato #393: IP address of wi-fi connect devices not showing on #status-devices.asp page?

wkearney99, 2018-02-07:
Is there an option or way to have the IP address of Wi-Fi connected devices shown on the status page?

http://192.168.1.2/#status-devices.asp

The MAC address shows, but the IP address column next to each of them is blank.  These being devices connecting via eth1 and eth2 (2.4 and 5 ghz radios, respectively).  My computer accessing it is showing an IP via br0 (the LAN uplink).  If I ping the router from other computers on the LAN their MAC and IP also show up on the device list.   

It's just the locally wi-fi connected ones that aren't showing their IP.  Even if I ping them from other hosts on the LAN, the device page does not update to show their IP address next to their MAC.

I'm running my R7000 using "Tomato Firmware 1.28.0000 -3.5-140 K26ARM USB AIO-64K"

I have this configured as just an Access Point.  I'm not using DNS or DHCP on this router itself.  I have DNS and DHCP being handled by a pfsense router elsewhere on the network.  This is connected via a LAN port (the WAN port is disabled and unused).

I'm guessing the display of the IP address is based on DHCP also running on the router?  Or should it be able to show this?

wkearney99, 2018-02-07:
I'm also noticing that dhcp-authoritative is set in /etc/resolv.dnsmasq.conf

I'm not running DHCP on this router, but on a pfsense router on the connected LAN.  I have DHCP shown as disabled on the br0 LAN network.

Is there a way to specifically disable the DHCP services via the web UI?  Or at least configure this as a DHCP relay?  Devices are getting a DHCP lease via the br0 LAN, so that's working.

malderhout, 2018-02-17:
Same issue here. Also on the R7000 with the latest advanced tomato firmware.