Jackysi/advancedtomato #394: Shibby dead? Sort of taken over by this new fork?

NoLooseEnds, 2018-02-09:
Hi, it looks like Shibby is not able to update as often as he would like (or have quit all together?), and that it is taken over by a fork by @kille72 (maybe he can elaborate?)

See: http://www.linksysinfo.org/index.php?threads/fork-tomato-by-shibby-compiled-by-kille72.73397/

Anyhow, maybe you should incorporate his builds from now on? Or even better, merge the two projects together.

Thanks.

Jackysi, 2018-02-09:
No. Kille72 decided to go his own way. I'm not changing forks. Specially because there are no AC/RT builds.

closed 2018-02-09 by Jackysi

NoLooseEnds, 2018-02-09:
Ok, fair enough. Do you know whats happening with Shibby's fork?

Jackysi, 2018-02-09:
Well last commit was 4th April. However Shibby is type of person who updates fork when he's done with release. So it could be that he has ready new update but hes not releasing it. Honestly this project is very time consuming and I don't have time either to do big updates. If I had time I'd merge various changes and create my own fork in both GUI and System development. But I just have no time.

I'm working for few companies and I also learn Golang in my free time. So i'm just overworked totally.

sha-265, 2018-02-22:
It's probably a good idea to merge between projects. I think it will be better if the few people that are working on these projects will cooperate, especially if they have no time.

I can't see any advantages for working (or almost not working) on two separated projects.