Jackysi/advancedtomato #398: Compilation of advancedtomato-arm

rejsmont, 2018-02-24:
Hi,

I am trying to compile advancedtomato-arm (need to compile some additional kernel modules). My dev environment is setup exactly like in this github repo (https://github.com/gavioto/advancedtomato-arm). Unfortunately, compilation of v140 fails with the following error:

```
$ make V1=1.28 V2=3.5-140 r7000z
```
```
init.o: In function `sysinit':
init.c:(.text.sysinit+0x2c44): undefined reference to `wl_defaults'
init.c:(.text.sysinit+0x2c48): undefined reference to `bsd_defaults'
init.c:(.text.sysinit+0x2c4c): undefined reference to `set_bcm4360ac_vars'
init.c:(.text.sysinit+0x2ec0): undefined reference to `wl_defaults'
init.c:(.text.sysinit+0x2ec4): undefined reference to `bsd_defaults'
init.c:(.text.sysinit+0x2ec8): undefined reference to `set_r8000_vars'
collect2: ld returned 1 exit status
```

Could you please provide some info on proper dev environment to compile?

Thanks in advance!

Radek

Jackysi, 2018-02-24:
https://advancedtomato.com/faq#compile-packages

rejsmont, 2018-02-24:
Hi!

Thank you for info! I have followed info from FAQ, but I am still getting the same compilation error. Here is my whole command history starting with clean ubuntu 16.04 x64:

```
$ sudo apt-get update
$ sudo apt-get install build-essential autoconf m4 bison flex g++ libtool sqlite gcc g++ binutils patch bzip2 make gettext unzip zlib1g-dev libc6 gperf sudo automake groff lib32stdc++6 libncurses5 libncurses5-dev gawk gitk zlib1g-dev autopoint shtool autogen mtd-utils gcc-multilib gconf-editor lib32z1-dev pkg-config libssl-dev automake1.11 libxml2-dev intltool libglib2.0-dev libstdc++5 texinfo dos2unix xsltproc libnfnetlink0 libcurl4-openssl-dev libxml2-dev libgtk2.0-dev libnotify-dev libevent-dev mc git texlive nettle-dev nodejs nodejs-legacy
$ sudo dpkg --add-architecture i386
$ sudo apt-get update
$ sudo apt-get install libelf1 libelf-dev:i386 libelf1:i386
$ sudo apt upgrade
$ sudo apt-get remove libicu-dev
$ wget https://advancedtomato.com/data/tools/automake_1.13.2-1ubuntu1_all.deb
$ wget https://advancedtomato.com/data/tools/automake_1.14.1-2ubuntu1_all.deb
$ wget https://advancedtomato.com/data/tools/libbison-dev_2.7.1.dfsg-1_amd64.deb
$ wget https://advancedtomato.com/data/tools/bison_2.7.1.dfsg-1_amd64.deb
$ sudo apt-get remove bison libbison-dev
$ sudo dpkg -i automake_1.13.2-1ubuntu1_all.deb
$ sudo dpkg -i automake_1.14.1-2ubuntu1_all.deb
$ sudo dpkg -i libbison-dev_2.7.1.dfsg-1_amd64.deb
$ sudo dpkg -i bison_2.7.1.dfsg-1_amd64.deb
$ git clone https://github.com/Jackysi/advancedtomato-arm.git
$ cd advancedtomato-arm/
$ git checkout v140
$ sudo ln -s release/src-rt-6.x.4708/toolchains/hndtools-arm-linux-2.6.36-uclibc-4.5.3 /opt/hndtools-arm-linux-2.6.36-uclibc-4.5.3
$ echo "export PATH=$PATH:/opt/hndtools-arm-linux-2.6.36-uclibc-4.5.3/bin:/sbin/" >> ~/.profile && source ~/.profile
$ cd advancedtomato-arm/release/src-rt-6.x.4708
$ make r7000z V1=1.28.0000 V2=3.5-140
```

rejsmont, 2018-02-25:
Thanks to AndreDVJ (http://www.linksysinfo.org/index.php?threads/compiling-advancedtomato-arm-v140.74053/) I have found out the reason. Apparently v140 tag was associated with arm7 branch. It would be great if releases in this repo were tagged on both branches (arm-v140 and arm7-v140) so users can pull the tag they want for the architecture they want.

Jackysi, 2018-05-10:
Awesome thanks!

closed 2018-05-10 by Jackysi