Jackysi/advancedtomato #400: OpenVPN Client Custom Configuration lost after submit

bjmi, 2018-03-10:
OpenVPN Client > Client 1 > Advanced > Custom Configuration
field is filled with e.g.
```
; pull "route-gateway "
; route-gateway 192.168.1.1

compress lz4
verify-x509-name 'CN=myhost.mydomain.com'
script-security 2
ca /jffs/ca.crt
up "/jffs/openvpn-tap.sh up"
down "/jffs/openvpn-tap.sh down"
```
After submitting the form _Custom Configuration_ is empty.
Result of `nvram get vpn_client1_custom` is empty too.

I used this configuration for a long time and overcomes several AT releases, maybe latest AT release contains a bug.

Router model: Linksys E3200

Jackysi, 2018-06-18:
Can you check the nvram usage? Could be full...

bjmi, 2018-07-14:
Maybe but
1. if this command is used at command line
```
nvram set vpn_client1_custom="; pull \"route-gateway\"
; route-gateway 192.168.1.1

compress lz4
ca \"/jffs/ca.crt\"
verify-x509-name 'CN=myhost.mydomain.com'
script-security 2
up \"/jffs/openvpn-tap.sh up\"
down \"/jffs/openvpn-tap.sh down\"
"
```
or
2. if preceding semi-colon of the first two lines is omitted

everything works as expected. It could be the semi-colon that causes this behavoir.