Jackysi/advancedtomato #402: WhatsApp Video and Call not working

G0n3D, 2018-04-10:
Hi here,
I have the latest version of Advancedtomato =>  Version: 3.5-140 on R7000

I have issue with WhatsApp application on my iPhone in Wifi.
The video call and voice call not working.

Regards,

Jackysi, 2018-06-18:
Sincerely doubt that's Tomato's fault in any way.

closed 2018-06-18 by Jackysi

G0n3D, 2018-06-18:
You kidding me ? i have try with another fw and another router and work good...
I think the problem is on this section:
[image]
Please reopen the issue....

Jackysi, 2018-06-18:
You just contridicted your self. Another fw/router... AdvancedTomato is gui modification of Tomato by Shibby. If you can prove it works on Tomato by Shibby version 140 on same router then I can re-open it. Otherwise Linksysinfo forums, not AT issues tracker (GUI only!!!!)