Jackysi/advancedtomato #406: Possible new build source with updates

Desani, 2018-04-17:
Hey Jacky,

I have recently been made aware of a Shibby Tomato fork called Fresh Tomato by kille72. Basically picking up where shibby left off, and adding a bunch of much needed security updates.

https://www.linksysinfo.org/index.php?threads/fork-freshtomato-arm.74117/

Would you consider putting the Advanced Tomato front end on this updated build? I refuse to use anything if it does not support Advanced Tomato.

Thanks,

Natteverf, 2018-05-02:
I second that!

april, 2018-05-04:
This gets my vote as well, as shibby is looking solidly like abandonware at this point.

broizter, 2018-05-26:
Here's the same fork but for MIPS devices http://www.linksysinfo.org/index.php?threads/fork-freshtomato-mips.74145/

LevizX, 2018-06-10:
Good idea, since Shibby won't be updating his repo any time soon, new source is warranted.

Jackysi, 2018-06-18:
True. But adapting to another firmware with so many changes is way too much for me at the time. Its A LOT of work. Also merging is almost impossible (attempted) without real manual work. So this is big process. 

When I have time. If ever that much. Otherwise help from community would be good (e.g. not having to work on this my self)

ecc29, 2018-06-21:
At this moment FreshTomato is basically the same GUI as the last version of Shibby Tomato. They just forked from Shibby Tomato and put some security updates. I guess AdvancedTomato doesn't need to start from scratch. I heard Shibby is very busy because he has a new baby. So we cannot expect any updates for Shibby Tomato in the near future. I really hope AdvancedTomato will not go to obsolete because of this.

botho, 2018-06-21:
+1
+1
+1

radarlog, 2018-07-03:
agree

FredericMa, 2018-08-01:
+1, would be very nice!

nemchik, 2018-08-12:
I've added a :thumbup: to the OP up top like many others have already, and hit the subscribe button to be notified of updates here. Hopefully others can do the same to express interest and keep the "me too" and "+1" comments to a minimum.

@Jackysi has labeled this issue which seems to mean it's not off the table. I'm sure he will update us if/when this gains traction.

HydrelioxGitHub, 2018-08-14:
Same thumb up here ...

hobbit19, 2018-11-28:
pls update firmare

boktai1000, 2018-12-01:
I've noticed there are two different repositories / forks for FreshTomato, @kille72 and pedro. Which one would this be built off of, @kille72 I'm assuming?

boktai1000, 2018-12-01:
@adriel oh I think you may have misunderstood my question, this issue is regarding building Advanced Tomato off a new base using FreshTomato instead of Tomato by Shibby. For FreshTomato, it seems there's two different versions/forks and I was just wondering which one of those two (kille72 or pedro) would be recommended to serve as the basis?

Maybe it doesn't matter - I'm not sure! Just something I was wondering about, as I imagine there are some slight differences between them. I think I'm getting ahead of myself a little bit though, if so my apologies! :)

adriel, 2018-12-01:
@boktai1000 ooh oops, yeah that makes more sense, my bad.

I always thought pedro and kille72 were working on the same one?
That's what it looked like [here](https://www.linksysinfo.org/index.php?threads/fork-freshtomato-arm.74117/) "Forked off from Tomato-ARM by Shibby, builds compiled by FreshTomato team
FreshTomato team: @kille72 @pedro311"

I could be wrong though.

PS, I've been running this fork of FreshTomato for a few weeks now and it seems pretty rock solid, but I do really miss the clean UI of AdvancedTomato :(

ecc29, 2018-12-21:
@boktai1000, this thread has been here for more than 8 month. I doubt if @Jackysi will read it (if he has time). Maybe you can open a new issue to talk about your discovery, especially for the new fork made by AndreDVJ. In case this project is abandoned, at least people will know where to find alternatives.

boktai1000, 2018-12-21:
@ecc29 done I have compiled what information I know of these builds as well as FreshTomato here: https://github.com/Jackysi/advancedtomato/issues/430 

Some users may not be aware of this if they only visit the Advanced Tomato website and maybe even this GitHub - I did not become aware of it until I did some digging in the comments and throughout the greater Tomato community.

Jackysi, 2018-12-22:
Actually I read every comment. I just refuse to respond to them. Very quick google search or even reading previous comments explains situation. I don't have life time of time to waste on responding to hundreds of emails/comments/messages on forums. I understand people want answers, but they're available by search. If they don't care enough to search then why should I to answer?

https://github.com/Jackysi/advancedtomato/issues/430