Jackysi/advancedtomato #415: Linksys WRT54GL not listed in Downloads (website)

bittner, 2018-07-05:
Is there a reason why the [Linksys WRT54GL](https://duckduckgo.com/?q=Linksys-WRT54GL+tomato&t=canonical&ia=products) is not listed on the [Downloads page](https://advancedtomato.com/downloads) of the website?

bittner, 2019-01-25:
Hello? Anyone home? :grimacing:

boktai1000, 2019-02-07:
Hi @bittner 

Not sure if you saw some of the other issues regarding the current state of AdvancedTomato, but the firmware and repository owner has other current real life priorities right now. In the background, a new (different) firmware has been gaining traction called FreshTomato. If you want to read up more on it, do a bit of Googling.

- #406
- #430 

Unfortunately, neither the AndreDVJ AdvancedTomato fork or FreshTomato support the WRT54GL as far as I can tell. But you're barking up the wrong tree here since the project is effectively on hiatus. I would recommend personally looking at another custom firmware such as the tried and true DD-WRT that has had years of support on it, but really that is another topic entirely that doesn't really belong on this GitHub issue tracking, but hopefully this information will get you up to speed with the state of things on AdvancedTomato as it stands.

closed 2020-04-23 by Jackysi