Jackysi/advancedtomato #416: The firmware is not compatiable with your router. Error code -1

ECTheRedPill, 2018-07-14:
I am trying to flash my:
**Netgear Nighthawk X6 R8000**
Hardware Version | R8000
Firmware Version | V1.0.4.18_10.1.49
GUI Language Version | V1.0.4.18_2.1.49.1
with:
https://advancedtomato.com/get/v3.5-140/K26ARM7/tomato-R8000-1.28.AT-ARM7-3.5-140-initial-64K.chk
I keep getting:|
The firmware is not compatiable (compatible - spelling error) with your router. Error code -1

Any help would be greatly appreciated.

rigid, 2019-01-18:
I got the same issue here. I tried to flash the image via tftp and get the following error:

```
CFE> tftpd
Start TFTP server
Reading :: Done. 5988410 bytes read
Not allow to downgrade firmware V1.0.2.99_1.1.99!
Reading ::
```
The CFE identifies itself as:

```
CFE for Foxconn Router R8000 version: v1.0.9
Build Date: Fri Dec 22 17:43:26 CST 2017
```

I'm not sure if this is the right place or if it's a pl_shibby/tomato-arm issue since it's unclear who compiles the initial .chk image and who can change the embedded version information.

Because the .chk image is meant to overwrite the stock firmware, I'd suggest to set the image version to something sensible that's always newer than any stock firmware of the future. (Until Netgear brings us yet another "security protection feature".)

rigid, 2019-01-18:
I got the same issue here. I tried to flash the image via tftp and get the following error:

```
CFE> tftpd
Start TFTP server
Reading :: Done. 5988410 bytes read
Not allow to downgrade firmware V1.0.2.99_1.1.99!
Reading ::
```
The CFE identifies itself as:

```
CFE for Foxconn Router R8000 version: v1.0.9
Build Date: Fri Dec 22 17:43:26 CST 2017
```

I'm not sure if this is the right place or if it's a pl_shibby/tomato-arm issue since it's unclear who compiles the initial .chk image and who can change the embedded version information.

Because the .chk image is meant to overwrite the stock firmware, I'd suggest to set the image version to something sensible that's always newer than any stock firmware of the future. (Until Netgear brings us yet another "security protection feature".)

EDIT: I tried to upload the image with the version bytes patched to a newer version than the currently installed stock firmware (which is V1.0.4.4_1.1.42) and I still get:
```
Not allow to downgrade firmware V1.0.5.99_1.2.99!
```