Jackysi/advancedtomato #438: how to set a lan port as trunk mode?

wood350, 2019-08-03:
my router is wnr3500lv2, firmware is adv tomato 1.40. I want to use the router as a vlan switch,and set lan port3 as trunk mode, I could not found a trunk related setting on vlan page, I guess log in telnet console can do it, but I can't found any usefull guide.any one can help me ?

wood350, 2019-08-03:
I setted two vlan groups, the vlan id is 1 and 2, they did not cotain port3, I just wanto set port3 as trunk mode, and use it accessing  both vlan 1 and vlan 2,how to make it ?

jurijp, 2020-09-23:
There is at least one reddit discussion about this problem here 
https://www.reddit.com/r/TomatoFTW/comments/5trbxd/having_trouble_with_advanced_tomato_vlan/ 
part of which points to a link here
There is a documented problem with Shibby's Tomato UI and the setting of VLANS 
https://bitbucket.org/pl_shibby/tomato-arm/issues/94/assigning-ports-to-vlans-doesnt-work
where the problem is dealt with by connecting to the router by ssh (ssh username@router_ip_address in Linux or Mac) or use putty for Windows. 
Thank goodness for the Wayback machine as the original doodleboobbers.com link referred to in the bitbucket has disappeared.  
http://web.archive.org/web/20180825195255/http://www.doodlebobbers.com/setting-vlan-ports-on-tomato-via-command-line/ for the sake of completeness I attach a pdf of that site
[Setting VLAN Ports on Tomato via Command Line » Doodlebobbers.pdf](https://github.com/Jackysi/advancedtomato/files/5265378/Setting.VLAN.Ports.on.Tomato.via.Command.Line.Doodlebobbers.pdf)
[Need help understanding tomato flashing and NVRAM - TomatoUSB.pdf](https://github.com/Jackysi/advancedtomato/files/5265379/Need.help.understanding.tomato.flashing.and.NVRAM.-.TomatoUSB.pdf)

I also attach a file sort of describing nvram commands which you will probably need and finally here's another person dealing with problems of VLAN setting with the GUI in Tomato.  https://www.scribblingbrain.com/2013/10/23/setting-up-tomatousb-for-vlans-and-ubiquiti-ap 
For about 2 years I have been trying to make vlans work with tomato and a trunk port from pfsense to no avail. I still have not accomplished it since when I set the various LANS, the Wifi networks I try to connect to just disconnect.  Let me know if you are able to figure this out, In fact I would love to understand how any of the posts in using VLANs for tomato ever work properly..