Jackysi/advancedtomato #439: Linksys WRT320N Bricked

RocketOnMC, 2019-09-29:
Hello,

About a year ago, I made the mistake of flashing the AdvancedTomato firmware to my Linksys WRT320N Router. It is no longer working. When I try turning on the router, the Power LED starts blinking. There is no default gateway and there is no internet access (obviously). Please advise me on what steps I can take to unbrick the router, if any.

Thanks!

jaysirod, 2020-04-16:
Not sure if you still need this, but reference [DD-WRT Unbricking Guide](https://wiki.dd-wrt.com/wiki/index.php/Recover_from_a_Bad_Flash), specifically _[the Peacock Thread](https://forum.dd-wrt.com/phpBB2/viewtopic.php?t=51486)_

RocketOnMC, 2020-04-17:
Hello again!
I went to the referenced Peacock Thread, and there is a linked article (https://forum.dd-wrt.com/phpBB2/viewtopic.php?t=63004), which tells me to 

> ....we all know the problems where wrt320n gets bricked bacause of full nvram (mostly openvpn versions because of all certificates) and due to faulty reset button (wired to wrong gpio) and faulty boot command it is impossible to debrick without serial console.

Now I found a way to do it using WPS button.

Please as soon as possible set:

nvram set reset_gpio=5
nvram commit

Don't forget to do it again if you restore defaults.

Or upgrade to 13493 build as soon as possible, it will set this automatically. You can get 13493 here: ftp://ftp.dd-wrt.com/others/eko/V24-K26/svn13491-snow/Linksys/WRT320N/

When this is set, you can press WPS button, plug power and keep it pressed for 10 - 12s. This will erase nvram and allow dd-wrt to boot again < 

Would this resolve my issue? If so, how would I go about doing this (I am a little bit confused on how exactly to do so).

Thanks once again!

jaysirod, 2020-04-17:
You are basically performing the 30/30/30 Reset on the router using WPS what it sounds like by the thread you posted. Also given the issues you mentioned, you most likely have Corrupted Firmware, go back to the peacock thread and follow the directions exactly on section 6. **" Is your router bricked?"**

Best of luck!

RocketOnMC, 2020-04-17:
After doing the 30/30/30 reset with the WPS button, I got new results. Unlike before, when I typed in "ping -t 192.168.1.1" into Powershell, it gave me a different IP, this being 192.168.1.19. After pinging 192.168.1.19, I would get a response. I attempted to access 192.168.1.19 using my browser (Chrome) and it gave me an error page. I then tried using Linksys's TFTP Utility along with the default firmware for my router, and I got an error, saying that the tool was "Unable to get responses from the server."
Could you please advise?

jaysirod, 2020-04-17:
Sure! 
What you first did was reset the firmware to "factory/stock settings" meaning any and all settings were erased. Following the Peacock Thread, 192.168.1.1 is a _dummy_ or placeholder for your real IP, try and find what your router's IP address is. 

From there following the Peacock Thread:
> If reply has TTL of 100, the bootloader (CFE) is responding. This is the best time to start the TFTP transfer. In most cases you should be able to flash dd-wrt firmware if you are getting any ttl=100 responses, as long as you flash at the start of these ping responses. See note 11 about how to flash. Timing can be tricky.
> 
> If reply has TTL of 64, the operating system firmware(i.e. Linux, dd-wrt) is responding. The good news here is that there IS operating system firmware on your router. 

Have you check the TTL (Time-To-Live) response from your router?

RocketOnMC, 2020-04-17:
Thanks!

I tried doing another 30/30/30 reset using both the reset button and the WPS button while leaving the ethernet plugged into the computer. 

I have checked the TTL response from my router, but whenever I type in "ping -t 192.168.1.1," I get "Destination host unreachable." I even tried setting the IP to the same static subnet IP as my router (so in my case, 192.168.1.10). Is there any other way I'm supposed to be getting the TTL response?

RocketOnMC, 2020-04-17:
Would [this](https://forum.dd-wrt.com/phpBB2/viewtopic.php?t=63444&start=15) work?

jaysirod, 2020-04-17:
I say give it a shot.

But instead of doing the 30/30/30 reset, just plugin the router, set a static ip address of your computer that's connected to router, make sure its on the same subnet then try pinging.

RocketOnMC, 2020-04-17:
Alright, will let you know.

RocketOnMC, 2020-04-17:
I tried that, and did not work. Pinging it gives "request time out" and "destination host unreachable errors following each other in Powershell.

jaysirod, 2020-04-17:
Do you have any signs of life on the router? (ie lights)

RocketOnMC, 2020-04-17:
Yes. First off, when I boot, all the lights (except the power light) turns green then turns off (which is normal for that model of router). As I type this out, the power light is blinking blue continuously, while LAN Port 1 (where I have my computer plugged in) is green, but blinking briefly every so often.

jaysirod, 2020-04-17:
Have you attempted to reboot (unplug-replug) while pinging continuously? I believe you have limited amount of time before the TFTP agent "dies"

RocketOnMC, 2020-04-17:
Yes I have, using the Powershell command, "ping -t 192.168.1.1." I still get the "Destination host unreachable" and the "Request timed out" errors throughout the process.

RocketOnMC, 2020-04-17:
Interestingly enough, after ending my most recent ping request (using ping -t), Powershell says that out of the 136 packets sent, 18 were received while 118 were lost. Is that a hopeful sign?

jaysirod, 2020-04-17:
Possibly, did you ensure that ip address of the router is correct?

RocketOnMC, 2020-04-17:
Yes. Before I flashed to the Tomato firmware that bricked my router, I do not remember the exact IP. I believe I had still kept it in within the 192.168.1.1[x] range.

jaysirod, 2020-04-17:
You can try a _for_ loop to iterate through all possible ip address:
`For ($i=0; $i -le 255; $i++) {
    ping 192.168.1.$1
    }`

I have one final guess is to attempt to connect blindly by getting a TFTP client, flash stock/oem to 192.168.1.1x (your routers ip) and hoping it connects.

RocketOnMC, 2020-04-17:
How do I know if I've hit the correct IP (it seems to be working)?

jaysirod, 2020-04-17:
You'll either get one of the responses
X = is the router ip address
NN = some number

> Reply from 192.168.1.X: bytes=NN time=NNms TTL=100
This means it's likely in TFTP mode

> Reply from 192.168.1.X: bytes=NN time=NNms TTL=64
This means it has an OS installed

> Request timed out
If there are no ttl= responses, do a hard reset on your router. ~ Peakcock

Check out the peacock thread again to see if any information is available to you.

RocketOnMC, 2020-04-17:
I am getting "transmit failure" all across the board (I stopped giving my adapter a static IP). How would I go about hard resetting my router since the 30/30/30 method doesn't work?

jaysirod, 2020-04-17:
You'll need to leave your adapter on the static network.

Try WPS method and seee the Peacock Thread Section 6 again, specifically perform steps a. through i.

RocketOnMC, 2020-04-17:
Will do. Also, how do I send commands like "nvram erase" to my router using PuTTY?

jaysirod, 2020-04-17:
nvram erase is used when the router is in normal operation. You'll need to use a TFTP client to "fully" recover your router.

RocketOnMC, 2020-04-17:
Hmm, got it. I read somewhere that using PuTTY, connecting the PC to the WAN port and then connecting it using the Serial option might just do the trick..

jaysirod, 2020-04-17:
You can try doing the Serial Trick, that is honestly a last resort if you need be as i believe it requires you to get into the router's motherboard

RocketOnMC, 2020-04-17:
I do have access to the Mobo right now. What steps would I take to ensure that the "Serial trick" works?

jaysirod, 2020-04-17:
See [Serial Recovery DD-WRT Wiki](https://wiki.dd-wrt.com/wiki/index.php/Serial_Recovery)