Jackysi/advancedtomato #443: dnsmasq custom configuration limited to 2048 characters ?

zjgithub, 2019-12-12:
Firstly thank you all for the great works. 

I flashed Asus RT-AC68U (RT-AC68U-AT-ARM-3.5-140-AIO-64K.trx) without problem, but when it comes to adding customer configuration for dnsmasq,  the webui gives error saying "limited to 2048 characters".

1. Can the limit be modified manually?
2.  If yes, what is the best way to do it ?

I hope this bug can be fixed in next build, because dnsmaq is very powerful and 2048 character limit seems really limiting its potential.

For now, I simply modified /etc/dnsmasq directly, but I'm not sure whether my manual intervention would break anything or not.

Thank you !

zjgithub, 2019-12-12:
hmm, I just rebooted my router, and noticed that my manual modifications to /etc/dnsmasq are all gone.  So I guess manual changes to /etc/dnsmasq is not sticky. I need to save it to nvram variable dnsmasq_custom ...

rampageX, 2019-12-12:
dnsmasq webui just need one line:
`conf-dir=/opt/etc/dnsmasq/custom/`

Then put all your custom config files of dnsmasq into /opt/etc/dnsmasq/custom/, no limited.

zjgithub, 2019-12-12:
That's wonderful ! Thank you !

Jackysi, 2020-04-23:
That is the best solution. Config limitation to 2048 lines is actually huge. The config is stored in NVRAM which is limited in size and when nvram gets full, it starts dumping out good data. I would say 2048 is safe and high limit, for extras use file system.

closed 2020-04-23 by Jackysi